enjoy-digital/litedram #317: Corresponding verilog testbench for ASIC

CarrolXC, 2023-01-04:
Hi, I generated a verilog module for the litedram core and now I want to simulate it as as ASIC design, basically I am trying to dummy out those interfaces with FPGA.  
Existed bench are all FPGA based.
Is that possible to get a corresponding testbench for the generated verilog? Or is there any reference or hints(like a python file testbench) I can take to write my own testbench?
Currently I only have the idea that I need to modfiy the python file to remove the FPGA part, it's a little tricky since I am not familiar with FPGA thus I cannot "draw" a clear boundary while truncating the design.
Sincerely appreciate any suggestions!!

enjoy-digital, 2023-01-10:
Hi @CarrolXC,

sorry, we don't yet have ASIC implementation of models to provides. You can see in the presentation [slides](https://docs.google.com/presentation/d/1mQOWqgmyQxpjLAzFwCulqgkp0TuxmaIDYp5iUfPqqIk/edit?usp=sharing)/P16 that the logic generated is specialized to the FPGA you are targeting, this the primitives you see integrated in the generated logic. There are some work planned for 2023 to add initial ASIC build backend support like we have actually for FPGAs; and this support will also allow to specialize the logic for the targeted ASIC flow.

So do to what you want, we don't yet have an out of the box solution, but you can either:
- 1) Generate with the current flow and replace the FPGA specific parts in the generated code by ASIC specific parts.
- 2) Create a custom build backend in LiteX and use it to specialize the code directly.

2) is the approach we will have to add initial ASIC support but required understanding a bit mode LiteX internals and logic specialization.

enjoy-digital, 2026-11-06:
Still current: LiteX has no ASIC backend today, so the two options above remain (replace the FPGA primitives in the generated code, or write a custom backend). For simulation only, `litedram_gen --sim` gives a core with no Xilinx primitives. Closing, feel free to re-open.

closed 2026-11-06 by enjoy-digital