enjoy-digital/litedram #325: --top-module 'sim' was not found in the design

CarrolXC, 2023-01-28:
I used the gen.py file to generate a litedram core, and meanwhile a build_xxxx.sh was generated as well.
When I tried to run build_xxx.sh file , it point to a make file in "litex" folder which call a top module as "sim". So it turned out an error since I don't have this file.
Do you have any any ideas on how this sim looks like , do I need to use the name "sim" when I try to generate the litedram core?
Appreciate it!
[image]
[image]

CarrolXC, 2023-01-28:
What I was trying to do is simulating the entire memory subsystem it seems like what I generated contains all the modules(not 100% sure). So the rest requirement need to be a testbench? Or what does the build_xxxx.sh file did actually?

CarrolXC, 2023-01-28:
What I was trying to do is simulating the entire memory subsystem and it seems like what I generated contains all the modules(not 100% sure). So the rest requirement might be a testbench? Do we need to manually build it ?What does the build_xxxx.sh file did actually?

enjoy-digital, 2023-01-31:
Hi @CarrolXC,

`litedram_gen` only generate the standalone core, not the test-bench. You'll have to create the test-bench yourself, with the simulator you are used to. `litedram_gen` allow you to integrate a memory model to the standalone core to simplify your simulations (`--sim`), this can be useful if don't want to integrate an official memory model, want to speed up the simulation (that's what we are using for example to test Linux in simulation).

If you want to re-integrate the standalone core with Verilator/LiteX, I would recommend looking at:
https://github.com/enjoy-digital/litex/wiki/Reuse-a-(System)Verilog,-VHDL,-(n)Migen,-Spinal-HDL,-Chisel-core
and some simulations:
https://github.com/enjoy-digital/litex_verilog_axi_test/blob/master/test_axi.py

CarrolXC, 2023-02-02:
@enjoy-digital Thanks for your explanation and suggestions! That's helpful.
I was able to call the /litedram/build/sim/gateware/sim.v by running /litedram/build/sim/gateware/run_sim.sh and find some memory-related results print in the terminal, it seems that the cpp file call the verilator to run this sim.v fille. So what is the purpose of this flow? Actually this folder is similar to what I generated from litedram_gen, both of them come with a build_sim.sh and run_sim.sh. The only difference is the name of ".v" file. Basically only in a folder containing a "sim.v" file could be run with run_sim.sh. What'the trick here. 
Sorry my question might sound childish since I am a totally newbie and trying to get some understanding on this whole system.
Appreciate your patience and help.
[image]

dinaabdelbaky, 2023-02-20:
@CarrolXC 
Hi Carrol, 
sorry to join the thread like this but, I am trying to use litedram core in simulation mode with my own testbench, same as what you are trying to do.
I have a couple of questions and hoping you might be able to help.
1-I want to confirm that I have generated the core successfully, I used the gen.py with the "--sim" and passed to it my own .yml file, after modifying one of the provided examples, is this it or should I have carried out more steps?
2-For the initialization in simulation, how did you manage to initialize the core, I checked the init_seq, and I tried to issue commands to the csr_reg yet no luck, any advice?
Thanks in advance!

CarrolXC, 2023-02-25:
Hi Dinaabdelbaky,
Actually I don't have enough ideas on your question but I am trying to
share my feeling.
Q1.The core you generated from gen.py should be a self-contained litedram
core(including phy if you add --sim) which you can directly use with proper
FPGA IP .
Q2.I was trying to write my own verilog to simulate the core, I am not sure
what the "init_seq" refers to, could you please point it out to me?
Overall this is a platform for FPGA applications which does not suit me
very well, I am still exploring. Good luck to you as well.
Thanks,
Xi

CarrolXC, 2023-02-25:
@dinaabdelbaky 
Hi Dina,
Actually I don't have enough ideas on your question but I am trying to
share my feeling.
Q1.The core you generated from gen.py should be a self-contained litedram
core(including phy if you add --sim) which you can directly use with proper
FPGA IP .
Q2.I was trying to write my own verilog to simulate the core, I am not sure
what the "init_seq" refers to, could you please point it out to me?
Overall this is a platform for FPGA applications which does not suit me
very well, I am still exploring. Good luck to you as well.
Thanks,

enjoy-digital, 2026-11-06:
Hi @CarrolXC,

sorry for the late follow-up. `litedram_gen --sim` generates a core to instantiate in your own testbench. It is not a complete Verilator simulation, but it also wrote a build script calling LiteX's Verilator Makefile, which expects a top module named `sim`, hence your error. #468 removes that script and its harness files from `--sim` outputs, so use `litedram_core.v` directly with your simulator.

closed 2026-11-06 by enjoy-digital