enjoy-digital/litedram #327: Carrying out the LiteDRAM standalone core initialization manually, through wishbone ctrl interface

dinaabdelbaky, 2023-02-28:
[image]
Hi,
I am new to the liteDRAM. I generated the core in a standalone mode, targeting Artix device, I also generated the corr. csv containing the csr addresses. I am using the core with Micron DDR3 model and a testbench to carryout come simulations and explorations of memory operations.
I checked out the generated sdram_phy.h it includes init_sequence, I tried to carry out the above sequence by writing to the corresponding cs registers. whoever I didn't see anything happening on the DFI interface or the memory interface.
Please point out what I am missing.
P.S: I have seen some similar questions, and some of the answers points out to writing data from an init file, but I don't really understand it, so please explain how to correctly initialize the core and the memory device.
Thanks in advance!

dinaabdelbaky, 2023-02-28:
Hi,
I am new to the liteDRAM. I generated the core in a standalone mode, targeting Artix device, I also generated the corr. csv containing the csr addresses. I am using the core with Micron DDR3 model and a testbench to carryout come simulations and explorations of memory operations.
I checked out the generated sdram_phy.h it includes init_sequence, I tried to carry out the above sequence by writing to the corresponding cs registers. whoever I didn't see anything happening on the DFI interface or the memory interface.
Please point out what I am missing.
P.S: I have seen some similar questions, and some of the answers points out to writing data from an init file, but I don't really understand it, so please explain how to correctly initialize the core and the memory device.
Thanks in advance!

yyx32, 2023-05-16:
Hi @dinaabdelbaky ，are you generated core with gen.py script?
Can you tell me how to get corr.csv and sdram_phy.h ?

dinaabdelbaky, 2023-05-18:
Hi @yyx32 Yes, I generated the core following those steps:
The addresses of the registers would be in the csv file.
The sdram_phy.h is generated under "output_drisoftware\include\generated"
[image]

dinaabdelbaky, 2023-05-18:
And I have worked out the issue I have mentioned above, the problem was not the sequence, I didn't notice that the adresses in the CSV were 32 bit wide and the wishbone interface was less wider. A simple fix to that is to take the address of the csr register in the generated csv and do right bit shifts, the amount of bit shift is equivalent 32-wishbone width

yyx32, 2023-05-24:
I will try it.Thanks bro.

gvillanovanm, 2025-02-22:
> And I have worked out the issue I have mentioned above, the problem was not the sequence, I didn't notice that the addresses in the CSV were 32 bit wide and the wishbone interface was less wide. A simple fix to that is to take the address of the csr register in the generated csv and do right bit shifts, the amount of bit shift is equivalent 32-wishbone width.

Nice @dinaabdelbaky ! So, were you able to use the DDR3 manually? Because I'm trying the same thing.

dinaabdelbaky, 2025-02-23:
@gvillanovanm Yes, I managed to get to work. I used the LiteDRAM core plus a DDR3 memory model available from micron as the DUT, and for the Testbench I used Cocotb. I hope this task is going smooth for you.

enjoy-digital, 2026-11-06:
Closing since this was solved above. For anyone landing here: the CSV addresses are byte addresses and `wb_ctrl_adr` is a word address, so shift them right by 2 (`sdram_dfii_pi0_command` at `0x1004` is `adr = 0x401`). Also note that `init_sequence()` does not include leveling; see #315.

closed 2026-11-06 by enjoy-digital