enjoy-digital/litedram #328: DDR4 reads without DQS at high speeds?

alexey-morozov, 2023-02-28:
I noticed that in your USDDRPHY design DQS is used for write only. As I understand it, you do this under an assumption that after DDR calibration procedure DQS read timing will have a fixed, stable relationship with the PHY clock. This seems to work with your current Component Mode-based implementation of the PHY which can run at speeds up to the US/USP specs limit of 1250 MT/s (625 MHz sys4x clock). The question is - will it still work well at higher frequencies e.g. when implemented with the use of Native Mode IO? According to my experiments, sys4x clock with Native Mode IO can be as high as 1120 MHz (2240 MT/s) on a xczu3cg device. Could it be that at such high frequencies DQS read jitter induced by the DDR chip itself could break the assumption?

Thanks in advance for your answer.

enjoy-digital, 2023-03-08:
Hi @alexey-morozov,

your understanding of the assumption is correct and the current implementation will indeed probably limit operation in native mode/high frequency This would need to be tested and eventually switching to use of DQS for read if required.

enjoy-digital, 2026-11-06:
The answer above still stands: the PHY still uses component mode with no DQS capture on reads, and nothing in its history adds native mode. Whether this holds at higher rates remains untested, and it isn't planned on the project side. Closing, happy to review a PR.