enjoy-digital/litedram #329: ulx3s example does not work

TheZoq2, 2023-04-01:
I'm trying to generate a litedram core for my ulx3s, but am having some trouble. As a first step, I want to see if I can get it working in the simplest configuration possible, so I removed all `user_ports` from the ulx3s example, modified the device from -45F to -85F as that's the device I have, and I changed the CPU to vexriscv instead of serv, to make the init process faster.
Finally, I changed the sys_clk_freq to 25e6 instead of 50e6, which is what the ulx3s clock is natively.

This is my modified yml

```yml
#
# This file is part of LiteDRAM.
#
# Copyright (c) 2021 Florent Kermarrec <florent@enjoy-digital.fr>
# SPDX-License-Identifier: BSD-2-Clause

{
    # General ------------------------------------------------------------------
    "device":  "LFE5U-85F-6BG381C", # FPGA device.
    "cpu":     "vexriscv",              # CPU type (ex vexriscv, serv, None)
    "memtype": "SDR",               # DRAM type
    "uart":    "rs232",             # Type of UART interface (rs232, fifo)

    # PHY ----------------------------------------------------------------------
    "sdram_module":    "IS42S16160", # SDRAM modules of the board or SO-DIMM
    "sdram_module_nb": 2,             # Number of byte groups
    "sdram_phy":       "GENSDRPHY",   # Type of FPGA PHY

    # Frequency ----------------------------------------------------------------
    "sys_clk_freq":     25e6,  # System clock frequency (SDR_clk = sys_clk)

    # Core ---------------------------------------------------------------------
    "cmd_buffer_depth": 16,    # Depth of the command buffer

    # User Ports ---------------------------------------------------------------
    "user_ports": {
    },
}
```

I built it with
```bash
python3 litedram/gen.py --name sdram_controller examples/ulx3s.yml
cd build/gateware && sh build_sdram_controller.sh
```

The core boots fine, but sdram_init fails 
```
litex> sdram_init
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 516918/524288
Memtest KO
```

I also tried changing the sdram_module to `IS42S16160` as that is what is included on pre-2022 ulx3s boards https://github.com/emard/ulx3s/blob/master/doc/MANUAL.md#board-versions

Finally I use this lpf file

```
BLOCK RESETPATHS;
BLOCK ASYNCPATHS;

# LOCATE COMP "ftdi_rxd" SITE "L4"; # FPGA transmits to ftdi
# LOCATE COMP "ftdi_txd" SITE "M1"; # FPGA receives from ftdi
LOCATE COMP "uart_tx" SITE "L4";
LOCATE COMP "uart_rx" SITE "M1";

LOCATE COMP "clk" SITE "G2";
IOBUF  PORT "clk" PULLMODE=NONE IO_TYPE=LVCMOS33;
FREQUENCY PORT "clk" 25 MHZ;

# Button 1
LOCATE COMP "rst" SITE "R1";  # FIRE1

# LOCATE COMP "sdram_clk" SITE "F19";
LOCATE COMP "user_clk" SITE "F19";
IOBUF PORT "user_clk" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
LOCATE COMP "user_rst" SITE "C10";

LOCATE COMP "sdram_cke" SITE "F20";
LOCATE COMP "sdram_cs_n" SITE "P20";
LOCATE COMP "sdram_we_n" SITE "T20";
LOCATE COMP "sdram_ras_n" SITE "R20";
LOCATE COMP "sdram_cas_n" SITE "T19";
LOCATE COMP "sdram_a[0]" SITE "M20";
LOCATE COMP "sdram_a[1]" SITE "M19";
LOCATE COMP "sdram_a[2]" SITE "L20";
LOCATE COMP "sdram_a[3]" SITE "L19";
LOCATE COMP "sdram_a[4]" SITE "K20";
LOCATE COMP "sdram_a[5]" SITE "K19";
LOCATE COMP "sdram_a[6]" SITE "K18";
LOCATE COMP "sdram_a[7]" SITE "J20";
LOCATE COMP "sdram_a[8]" SITE "J19";
LOCATE COMP "sdram_a[9]" SITE "H20";
LOCATE COMP "sdram_a[10]" SITE "N19";
LOCATE COMP "sdram_a[11]" SITE "G20";
LOCATE COMP "sdram_a[12]" SITE "G19";
LOCATE COMP "sdram_ba[0]" SITE "P19";
LOCATE COMP "sdram_ba[1]" SITE "N20";
LOCATE COMP "sdram_dm[0]" SITE "U19";
LOCATE COMP "sdram_dm[1]" SITE "E20";
LOCATE COMP "sdram_dq[0]" SITE "J16";
LOCATE COMP "sdram_dq[1]" SITE "L18";
LOCATE COMP "sdram_dq[2]" SITE "M18";
LOCATE COMP "sdram_dq[3]" SITE "N18";
LOCATE COMP "sdram_dq[4]" SITE "P18";
LOCATE COMP "sdram_dq[5]" SITE "T18";
LOCATE COMP "sdram_dq[6]" SITE "T17";
LOCATE COMP "sdram_dq[7]" SITE "U20";
LOCATE COMP "sdram_dq[8]" SITE "E19";
LOCATE COMP "sdram_dq[9]" SITE "D20";
LOCATE COMP "sdram_dq[10]" SITE "D19";
LOCATE COMP "sdram_dq[11]" SITE "C20";
LOCATE COMP "sdram_dq[12]" SITE "E18";
LOCATE COMP "sdram_dq[13]" SITE "F18";
LOCATE COMP "sdram_dq[14]" SITE "J18";
LOCATE COMP "sdram_dq[15]" SITE "J17";
IOBUF PORT "sdram_cke" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_cs_n" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_we_n" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_ras_n" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_cas_n" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[0]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[1]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[2]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[3]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[4]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[5]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[6]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[7]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[8]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[9]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[10]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[11]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_a[12]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_ba[0]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_ba[1]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dm[0]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dm[1]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[0]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[1]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[2]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[3]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[4]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[5]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[6]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[7]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[8]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[9]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[10]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[11]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[12]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[13]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[14]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "sdram_dq[15]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;

LOCATE COMP "init_done" SITE "H3";
LOCATE COMP "init_error" SITE "E1";
```
in which I copied most configurations from https://github.com/emard/ulx3s/blob/master/doc/constraints/ulx3s_v20.lpf

The only significant changes I made to it was to remove `sdram_clk` and replace it by `user_clk` (under the assumption that `user_clk` only forwards the `sys_clk`. I also renamed `sdram_dqm` from the original file with `sdram_dm`

Am I missing something here that makes this not work?

TheZoq2, 2023-04-03:
I finally got this to work by crossreferencing some stuff from https://github.com/BrunoLevy/learn-fpga

The core problem turned out to be that the dram module seems to need to run at 50MHz, and the dram clock must be pahse offset from the system clock

To make it work, for anyone reading:

Generate a dram controller
```
p3 litedram/gen.py --name sdram_controller examples/ulx3s.yml
```

Set up a custom `build.sh`, `lpf.lpf` and `build.ys`

build.sh:
```
set -e
yosys -l sdram_controller.rpt build.ys
nextpnr-ecp5 --json sdram_controller.json --lpf lpf.lpf --textcfg sdram_controller.config  --85k --package CABGA381 --speed 6 --timing-allow-fail --seed 1 
ecppack  --bootaddr 0   --compress  sdram_controller.config --svf build/gateware/sdram_controller.svf --bit sdram_controller.bit 
```
build.ys
```
verilog_defaults -push
verilog_defaults -add -defer
read_verilog /home/frans/build/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v
read_verilog /home/frans/Documents/fpga/litex/litedram/build/gateware/sdram_controller.v
read_verilog top.v
verilog_defaults -pop
attrmap -tocase keep -imap keep="true" keep=1 -imap keep="false" keep=0 -remove keep=0

synth_ecp5  -top top
write_json  sdram_controller.json
```
lpf.lpf
```
BLOCK RESETPATHS;
BLOCK ASYNCPATHS;

# LOCATE COMP "ftdi_rxd" SITE "L4"; # FPGA transmits to ftdi
# LOCATE COMP "ftdi_txd" SITE "M1"; # FPGA receives from ftdi
LOCATE COMP "uart_tx" SITE "L4";
LOCATE COMP "uart_rx" SITE "M1";

# Button 1
LOCATE COMP "rst" SITE "R1";  # FIRE1

# LOCATE COMP "sdram_clk" SITE "F19";
LOCATE COMP "user_clk" SITE "F19";
IOBUF PORT "user_clk" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4;
LOCATE COMP "user_rst" SITE "C10";

LOCATE COMP "clk25" SITE "G2";
IOBUF PORT "clk25" IO_TYPE=LVCMOS33;
LOCATE COMP "rst" SITE "R1";
IOBUF PORT "rst" IO_TYPE=LVCMOS33;
LOCATE COMP "sdram_clock" SITE "F19";
IOBUF PORT "sdram_clock" IO_TYPE=LVCMOS33;
LOCATE COMP "wifi_gpio0" SITE "F1";
IOBUF PORT "wifi_gpio0" IO_TYPE=LVCMOS33;
LOCATE COMP "serial_tx" SITE "L4";
IOBUF PORT "serial_tx" IO_TYPE=LVCMOS33;
LOCATE COMP "serial_rx" SITE "M1";
IOBUF PORT "serial_rx" IO_TYPE=LVCMOS33;
LOCATE COMP "sdram_a[0]" SITE "M20";
IOBUF PORT "sdram_a[0]" IO_TYPE=LVCMOS33;
IOBUF PORT "sdram_a[0]" SLEWRATE=FAST;
LOCATE COMP "sdram_a[1]" SITE "M19";
IOBUF PORT "sdram_a[1]" IO_TYPE=LVCMOS33;
IOBUF PORT "sdram_a[1]" SLEWRATE=FAST;
LOCATE COMP "sdram_a[2]" SITE "L20";
[... 185 lines cut ...]
        , .sdram_ras_n(sdram_ras_n)
        , .sdram_cas_n(sdram_cas_n)
        , .sdram_we_n(sdram_we_n)
        , .sdram_cs_n(sdram_cs_n)
        , .sdram_dm(sdram_dm)
        , .sdram_dq(sdram_dq)
        , .sdram_cke(sdram_cke)
        , .init_done(init_done)
        , .init_error(init_error)
        , .user_clk(user_clk)
        , .user_rst(user_rst)
        );

    assign sys_ps_clk = main_ecp5pll0_clkout1;
    ODDRX1F ODDRX1F(
        .D0(1'd1),
        .D1(1'd0),
        .SCLK(sys_ps_clk),
        .Q(sdram_clock)
    );

endmodule
```

Perhaps there are better ways to achieve this, but at least this gets it working.

enjoy-digital, 2023-04-24:
Hi @TheZoq2,

sorry for the delay and good to know you've got it running. The approach used seems fine. Thanks for sharing your project, this will be useful for users willing do the the same. I'll also probably add a link to it in the wiki.

enjoy-digital, 2026-11-06:
Hi @TheZoq2,

sorry for not following up on this. Your analysis was right, and thanks again for sharing the working setup. With `GENSDRPHY`, the generated core had no PLL and no SDRAM clock output: `sys_clk` came directly from `clk`, `user_clk` was the same clock, and the integrator had to generate `clk` and drive the SDRAM clock pin with a phase-shifted copy. That is what your `top.v` does. Forwarding `user_clk` directly gives no phase margin, which most likely explains your first memtest errors.

#467 now does this in the core on ECP5: with `input_clk_freq` in the YAML, a PLL generates `sys_clk`, and the core drives a new `sdram_clk` output with a phase set by `sdram_clk_phase` (default 90). `examples/ulx3s.yml` uses it (25 MHz input, 50 MHz `sys_clk`), so your `top.v` PLL and `ODDRX1F` are no longer needed: connect the board clock to `clk` and `sdram_clk` to pin F19. The generated PLL settings match the ones in your `top.v`. Both modes are described in `doc/gen.md`.

This was checked in generation and through Yosys and nextpnr on the ULX3S pins (timing met at 50 MHz), but not on a board. If you still have the setup, a test would be very welcome.

closed 2026-11-06 by enjoy-digital