enjoy-digital/litedram #331: Generate liteDRAM verilog file

ztachip, 2023-04-18:
I am trying to generate just the DRAM controller for the Arty board using command

./gen.sh ../examples/arty.yml

But the generated verilog also contains VexRiscv.

How can I generate just the DRAM controller?

Also what is the frequency that I input to the controller, is it "input_clk_freq"? Can I change it to different clock value?

Thanks

ztachip, 2023-04-18:
Ok. That makes sense about the CPU (-:
About the user_clk, I guess this is the clock for user AXI port? 
Is this clock the same as input_clk_freq? Do I have a choice on the user_clk frequency?
Thx

rowanG077, 2023-04-18:
From what I gather these can be the same. But can also be different.

ztachip, 2023-04-19:
I generate the standalone litedram for Arty7 using command... 
gen.py ../examples/arty.yml 
However, I don't see how VexRiscv referenced in the generated verilog being generated.
Can you help? Thx

ztachip, 2023-04-19:
I found it. I was in the python-cpu-vexrisc.
Thanks

closed 2023-04-23 by ztachip