enjoy-digital/litedram #333: Setting for user_clk

ztachip, 2023-04-23:
I assume that user_clk is the clock for the AXI interface.
Do I have a choice on this clock? Can I set user_clk to any clock value?
Thanks

enjoy-digital, 2023-04-24:
Hi @ztachip,

`user_clk` is indeed the clock of the user interfaces. This is provided by the core and you have to use it for the user interfaces. It's possible to adjust this clk by setting `sys_clk_freq` but I would  first recommend using a value closed to the one used in an example to start with and then do small change/test increments.

The variation range will be limited by:
- The frequency range of the PLL used to generate this clk from the primary clk.
- The frequency range of the DRAM.

enjoy-digital, 2026-11-06:
Closing as answered above: `user_clk` is set through `sys_clk_freq`, within the PLL and DRAM frequency limits. Feel free to re-open if needed.

closed 2026-11-06 by enjoy-digital