enjoy-digital/litedram #344: LiteDRAM DDR3 Core targeting Arty AXI read data appears on the Native port instead.

dinaabdelbaky, 2023-08-17:
Hi,
I have generated LiteDRAM core targeting DDR3 with the configuration in Fig1. I have noticed for a read command issued from the AXI interface the rd_data is returned in the rd_data of native interface, not the AXI interface Fig2 
[image]
[image]

Qiange516, 2024-04-25:
@dinaabdelbaky 
Hello, can you read and write AXI_Litedram normally?
 I encountered the same type of problem as  https://github.com/enjoy-digital/litedram/issues/342 （Axi port write data error #342）
If it's convenient, leave an email （bumianzhe@126.com）to communicate, thank you.

enjoy-digital, 2026-11-06:
Hi @dinaabdelbaky,

sorry for the very late answer. This is expected: the crossbar drives the controller's read data bus to the `rdata_data` of every port and only `rdata_valid` is per port. So the data of a read issued on the AXI port also shows up on `user_port_native_0_rdata_data`, without `user_port_native_0_rdata_valid`, while the AXI read itself is returned on the R channel with `rvalid`. `rdata_data` is only meaningful together with `rdata_valid`; this is now stated in the native port protocol section of `doc/gen.md` (#465).

Closing, feel free to re-open if an AXI read really does not come back on the R channel.

closed 2026-11-06 by enjoy-digital