enjoy-digital/litedram #355: Wishbone port does not accept more than one operation despite cmd_buffer_depth non-null in LiteDram yaml file,

fontamsoc, 2024-03-30:
In my LiteDram yaml file, I am using Wishbone user port as follow:
```
"user_ports": {
    "wishbone_0" : {
        "type": "wishbone",
    },
},
```

The generated Wishbone port does not have a `stall` signal, and asserting the `stb` signal more than once without waiting for the corresponding `ack` signal assertion, causes `ack` signal to never get asserted.

I have `cmd_buffer_depth` in the LiteDram yaml as follow:
```
"cmd_buffer_depth": 16, # Depth of the command buffer
```

Isn't the cmd_buffer_depth in LiteDram yaml file meant to allow more than one operation without waiting for the `ack` signal assertion ?

enjoy-digital, 2026-11-06:
Hi @fontamsoc,

sorry for the late answer. Yes, this is expected: the Wishbone user port of the standalone core (like every Wishbone interface in LiteX) is Wishbone **classic**, one transfer at a time: `cyc`/`stb` must stay high until the `ack` cycle and the next transfer can only start after it. There is no `stall` and no pipelined (B4) mode. The bridge behind the port only looks at `stb` while it waits for a command, so when `stb` is pulsed once per transfer, the pulses issued while the first transfer is in flight are simply lost and the master then waits forever for `ack`s that will never come, which is what you observed.

`cmd_buffer_depth` is not related to the Wishbone port: it is the depth of the controller's per-bank command queues, which only the native and AXI user ports (several commands in flight) can fill. This is now documented in [doc/gen.md, "Wishbone port protocol"](https://github.com/enjoy-digital/litedram/blob/doc/wishbone-port/doc/gen.md#wishbone-port-protocol), next to the native port protocol (#463).

For more throughput than one transfer per round trip: use the full port `data_width` (one transfer then moves a whole DRAM word, e.g. 128 bits for a x16 DDR3 at 1:4), an `axi` user port (several outstanding bursts; that's what I'd recommend for a pipelined master) or a `native` port with several commands in flight.

enjoy-digital, 2026-11-06:
Follow-up: #464 adds an opt-in Wishbone B4 pipelined mode to the standalone Wishbone port, which is what your master needs. With `"pipelined": true` on the port (and `"pipelined_depth"`, default 8), the generated port gets a `stall` output: a beat is accepted on every cycle where `cyc`/`stb` are high and `stall` is low, you can present the next beat right after, and the `ack`s come later in order, one per transfer, with up to `pipelined_depth` transfers in flight. Classic stays the default (the two modes can't share one slave: a classic master's `stb` held until `ack` would look like repeated beats to a pipelined slave). Checked in simulation only, so a report from a real master would be appreciated.