enjoy-digital/litedram #356: LiteDRAMDMAWriter sinks data when not enabled

DaveBerkeley, 2024-04-24:
The sink Stream in LiteDRAMDMAWriter has the 'ready' flag set by default.

If with_csr is set, shouldn't the 'sink.ready' flag only be set if the 'enable' bit is set in the CSR?

At the moment you can connect a Stream to the DMA and it will accept data even when the DMA is not enabled. This seems wrong.

I want to be able to set the 'base'/'length' of the transfer, pointing to an address in DRAM. Then set 'enable'. It would also be good to be able to use the sink Stream's 'last' flag to terminate the transfer and generate an event. So you can wait for a packet to complete, without having to know in advance the number of bytes to be transferred. The transfer should stop & generate an event when the 'last' word is transferred, or the 'offset' number of words have been transferred, whichever happens first.

The 'ready' logic can be fixed as follows : 

diff --git a/litedram/frontend/dma.py b/litedram/frontend/dma.py
index 3f31455..4caf37a 100644
--- a/litedram/frontend/dma.py
+++ b/litedram/frontend/dma.py
@@ -251,7 +251,7 @@ class LiteDRAMDMAWriter(Module, AutoCSR):
         self.submodules.fsm = fsm
         self.comb += fsm.reset.eq(~self._enable.storage)
         fsm.act("IDLE",
-            self.sink.ready.eq(1),
+            self.sink.ready.eq(0), # should only be enabled when on
             NextValue(offset, 0),
             NextState("RUN"),
         )
@@ -260,7 +260,7 @@ class LiteDRAMDMAWriter(Module, AutoCSR):
             self._sink.last.eq(offset == (length - 1)),
             self._sink.address.eq(base + offset),
             self._sink.data.eq(self.sink.data),
-            self.sink.ready.eq(self._sink.ready),
+            self.sink.ready.eq(self._sink.ready & self._enable.storage), # must be enabled
             If(self.sink.valid & self.sink.ready,
                 NextValue(offset, offset + 1),
                 If(self._sink.last,

DaveBerkeley, 2024-04-26:
I'm working on a pull-request for this

https://github.com/DaveBerkeley/litedram/blob/issues_356/litedram/frontend/dma.py

It works well with my Amaranth / Streams interface. But I need a way to check backward compatibility.

enjoy-digital, 2026-10-21:
Closing since #389 has been merged and there was no further feedback. Feel free to reopen if the issue is still seen.

closed 2026-10-21 by enjoy-digital