enjoy-digital/litedram #357: Adding support to gowin phys in litedram.gen?

linhz0hz, 2024-04-28:
It seems that gowin phys are included in the support but not included in litedram.gen. The main TODO seems to adding LiteDRAMGW*DDRPHYCRG ? I would like to help but I might now know enough.

enjoy-digital, 2024-06-19:
Hi @linhz0hz,

`GW2DDRPHY` could be added to litedram.gen probably without too much difficulty. The `GW2DDRPHY` in very similar to the `ECP5DDRPHY` that is already integrated in litedram.gen, so you could look at the approach used for ECP5.

The `GW5DDRPHY` is not yet finished so integration on litedram.gen could wait a bit.

ZhaoCake, 2024-10-22:
> Hi,
> 
> `GW2DDRPHY` could be added to litedram.gen probably without too much difficulty. The `GW2DDRPHY` in very similar to the `ECP5DDRPHY` that is already integrated in litedram.gen, so you could look at the approach used for ECP5.
> 
> The `GW5DDRPHY` is not yet finished so integration on litedram.gen could wait a bit.

We would like to help support GW5DDRPHY in litedram. How's this progress going these days? What should we do to support it, may you talk about this? Thank you for your help very much and thank you for the awesome project `litex`.

enjoy-digital, 2026-11-06:
Hi @linhz0hz,

an update on this: both Gowin PHYs are complete now (`GW2DDRPHY` at 1:2, `GW5DDRPHY` at 1:2 and 1:4, running on the Tang Primer 20K, Tang Mega 60K/138K/138K Pro, Tang Console and LJPI boards in LiteX-Boards), and #462 adds them to `litedram_gen`: a `LiteDRAMGowinDDRPHYCRG` (the TODO you had identified), the `GowinPlatform` selection, `devicename`/`sdram_rate`/`dll_off` YAML keys and two examples, `examples/tang_primer_20k.yml` (GW2A) and `examples/tang_mega_138k_pro.yml` (GW5A), documented in `doc/gen.md`.

The generation is checked for both examples and a 1:4 variant, but the standalone core has not been run on a board yet: if you (or anyone reading this) can give the generated core a try on a Gowin board, feedback is welcome.

closed 2026-11-06 by enjoy-digital